STMicroelectronics/stm32u0xx-hal-driver #2: Typo in cmsis definition usage

fpistm, 2024-08-28:
This macro includes a typo: `RCC_CR_HSIDY_Pos` shoudl be `RCC_CR_HSIRDY_Pos`
https://github.com/STMicroelectronics/stm32u0xx_hal_driver/blob/f949e9dd4d8509051e0e4fd34067a3cfc37fe089/Inc/stm32u0xx_hal_rcc.h#L622

Here the correct syntax:
https://github.com/STMicroelectronics/cmsis_device_u0/blob/7016f566979a3a25a02478033782c38f3f8999b6/Include/stm32u031xx.h#L4920
https://github.com/STMicroelectronics/cmsis_device_u0/blob/7016f566979a3a25a02478033782c38f3f8999b6/Include/stm32u073xx.h#L5427
https://github.com/STMicroelectronics/cmsis_device_u0/blob/7016f566979a3a25a02478033782c38f3f8999b6/Include/stm32u083xx.h#L5685

Workaround:
```C
#define RCC_CR_HSIDY_Pos RCC_CR_HSIRDY_Pos
```

TOUNSTM, 2024-08-30:
ST Internal Reference: 189876

KRASTM, 2024-09-18:
Fixed in commit https://github.com/STMicroelectronics/stm32u0xx_hal_driver/commit/531a52e4015b1985e75488af1cdb44ac3881e7da

closed 2024-09-18 by KRASTM