STMicroelectronics/stm32u0xx-hal-driver #3: Error in stm32u0xx_hal_pwr_ex.h documentation

blackbird998, 2024-12-18:
Hi,
Starting from the line 227 in ` stm32u0xx_hal_pwr_ex.h`
```c
/** @defgroup PWREx_Flag  PWR Status Flags
  *        Elements values convention: 0000 0000 0XXY YYYYb
  *           - Y YYYY  : Flag position in the XX register (5 bits)
  *           - XX  : Status register (2 bits)
  *                 - 01: SR1 register
  *                 - 10: SR2 register
  *        The only exception is PWR_FLAG_WU, encompassing all
  *        wake-up flags and set to PWR_SR1_WUF.
  * @{
  */
[...]
#define PWR_FLAG_WUFI                        0x010FU             /*!< Wakeup on internal wakeup line            */
#define PWR_FLAG_FLASHRDY                    0x0207U             /*!< flash ready flag                          */
[...]
```
the `XX` status register bits seem to represent bits 8-9 and `Y` only bits 0-3 (`0000 00XX 0000 YYYYb`) and not 5-6 and 0-4 respectively (`0000 0000 0XXY YYYYb`).

Kind regards,
Adam

KRASTM, 2024-12-20:
ST Internal Reference: 199231

RJMSTM, 2025-01-08:
Fixed in : c9b2daabb7ee3b69237f51c7110b5238781de3a4

closed 2025-01-08 by RJMSTM