STMicroelectronics/stm32u0xx-hal-driver #5: HAL_CRC_Accumulate interface throws warnings if used with 8 or 16 byte arrays.

scotteshaw, 2025-03-19:
The interface to `HAL_CRC_Accumulate` and `HAL_CRC_Calculate` both take `uint32_t pBuffer[]` no matter the input data format. I can see that this is not *technically* unsafe since it is cast back to a `uint8_t *` when the data format is byte, but it it ensures that warnings are thrown when `-Wcast-align` is used which is an extremely sensible thing to do when building for cortex M0 parts.

```
[1/2] Building CXX object applications/blinky/CMakeFiles/blinky.dir/blinky.cpp.obj
../../applications/blinky/blinky.cpp: In function 'uint16_t BLINKY::CalcHalCrc(uint8_t*, size_t)':
../../applications/blinky/blinky.cpp:179:40: warning: cast from 'uint8_t*' {aka 'unsigned char*'} to 'uint32_t*' {aka 'long unsigned int*'} increases required alignment of target type [-Wcast-align]
  179 |    crc = HAL_CRC_Calculate(&crcHandle, (uint32_t *)data, size);
```

Can this please be improved?

ALABSTM, 2025-03-24:
Hi @scotteshaw,

Thank you for this report. I tried building the **STM32U083C-DK**'s CRC [example](https://github.com/STMicroelectronics/STM32CubeU0/tree/main/Projects/STM32U083C-DK/Examples/CRC/CRC_Example) using STM32CubeIDE **1.16.0** with flags `-Wextra` and `-Wconversion` enabled but could **not** see warnings similar to those you reported. I'll give it another try.

Meanwhile, would you mind indicating the IDE you are using, or at least the compiler, and its version?

I also noticed from the log you shared that the warning is related to the `CalcHalCrc(uint8_t*, size_t)` function, which looks like a function you defined at your application's level. Could it just be that the first parameter be changed into a `uint32_t *`?

With regards,

scotteshaw, 2025-03-24:
Try with `-Wcast-align` this one isn't included in `-Wextra`.

I am using a cmake based project, happens to be in vs code.

GCC version: `arm-none-eabi-gcc (GNU Toolchain for the Arm Architecture 11.2-2022.02 (arm-11.14)) 11.2.1 20220111`

You are correct, I posted the output from a slightly earlier version of the code. Here is the updated output:

```
[1/2] Building CXX object applications/blinky/CMakeFiles/blinky.dir/blinky.cpp.obj
../../applications/blinky/blinky.cpp: In function 'uint16_t BLINKY::CalcHalCrc(uint8_t*, size_t)':
../../applications/blinky/blinky.cpp:179:40: warning: cast from 'uint8_t*' {aka 'unsigned char*'} to 'uint32_t*' {aka 'long unsigned int*'} increases required alignment of target type [-Wcast-align]
  179 |    crc = HAL_CRC_Calculate(&crcHandle, (uint32_t *)data, size);
      |                                        ^~~~~~~~~~~~~~~~
```

Here are some excerpts of the code I am playing with which I extracted from the proprietary codebase I am using.

```c
volatile uint16_t crcHal = 0;

uint16_t CalcHalCrc(uint8_t * data, size_t size);

void Task()
{
   if (timeSinceLastLedToggle_BlinkyTaskPeriods >= BLINKY_LED_TOGGLE_DELAY)
   {
      BLINKY_DRIVER::toggleLed();
      // Reset the delay tracker
      timeSinceLastLedToggle_BlinkyTaskPeriods = 0;

      uint8_t data[4] = {1, 2, 3, 4};

      crcHal = CalcHalCrc(data, 4);
   }
   else
   {
      timeSinceLastLedToggle_BlinkyTaskPeriods += 1;
   }
}

uint16_t CalcHalCrc(uint8_t * data, size_t size)
{
   uint32_t crc;

   CRC_HandleTypeDef crcHandle;
   crcHandle.Instance = CRC;
   crcHandle.Init.DefaultPolynomialUse = DEFAULT_POLYNOMIAL_DISABLE;
   crcHandle.Init.DefaultInitValueUse = DEFAULT_INIT_VALUE_DISABLE;
   crcHandle.Init.GeneratingPolynomial = 0x5935;
   crcHandle.Init.CRCLength = CRC_POLYLENGTH_16B;
   crcHandle.Init.InitValue = 0x0000;
   crcHandle.Init.InputDataInversionMode = CRC_INPUTDATA_INVERSION_NONE;
   crcHandle.Init.OutputDataInversionMode = CRC_OUTPUTDATA_INVERSION_DISABLE;
   crcHandle.InputDataFormat = CRC_INPUTDATA_FORMAT_BYTES;

   __HAL_RCC_CRC_CLK_ENABLE();
   __HAL_RCC_CRC_FORCE_RESET();
   __HAL_RCC_CRC_RELEASE_RESET();
   HAL_CRC_Init(&crcHandle);

   crc = HAL_CRC_Calculate(&crcHandle, (uint32_t *)data, size);
   HAL_CRC_DeInit(&crcHandle);
   __HAL_RCC_CRC_CLK_DISABLE();

   return static_cast<uint16_t>(crc);
}
```

ALABSTM, 2025-04-23:
Hi @scotteshaw,

I managed to reproduce the behavior you reported using using **STM32CubeIDE 1.16.0** with flag `-Wcast-align` enabled, and to get the warning message.

To get rid of this warning, I propose that you handle this at your **application's level**. A solution could be to use **unions**.

Here is how I modified the `main.c` file of the **STM32U083C-DK**'s CRC [example](https://github.com/STMicroelectronics/STM32CubeU0/tree/main/Projects/STM32U083C-DK/Examples/CRC/CRC_Example): 

```diff
/* Private define ------------------------------------------------------------*/

/* USER CODE BEGIN PD */
#define BUFFER_SIZE    114
/* USER CODE END PD */

+/* Private typedef -----------------------------------------------------------*/
+
+/* USER CODE BEGIN PTD */
+union alignedBuffer
+{
+  uint8_t  u08DataBuffer[BUFFER_SIZE * 4];
+  uint32_t u32DataBuffer[BUFFER_SIZE];
+};
+/* USER CODE END PTD */

/* Private variables ---------------------------------------------------------*/

CRC_HandleTypeDef hcrc;

/* USER CODE BEGIN PV */

/* Used for storing CRC Value */
__IO uint32_t uwCRCValue = 0;

-static const uint32_t aDataBuffer[BUFFER_SIZE] =
+static const union alignedBuffer aDataBuffer =
{
+  .u32DataBuffer =
+  {
  0x00001021, 0x20423063, 0x408450a5, 0x60c670e7, 0x9129a14a, 0xb16bc18c,
  0xd1ade1ce, 0xf1ef1231, 0x32732252, 0x52b54294, 0x72f762d6, 0x93398318,
  0xa35ad3bd, 0xc39cf3ff, 0xe3de2462, 0x34430420, 0x64e674c7, 0x44a45485,
  0xa56ab54b, 0x85289509, 0xf5cfc5ac, 0xd58d3653, 0x26721611, 0x063076d7,
  0x569546b4, 0xb75ba77a, 0x97198738, 0xf7dfe7fe, 0xc7bc48c4, 0x58e56886,
  0x78a70840, 0x18612802, 0xc9ccd9ed, 0xe98ef9af, 0x89489969, 0xa90ab92b,
  0x4ad47ab7, 0x6a961a71, 0x0a503a33, 0x2a12dbfd, 0xfbbfeb9e, 0x9b798b58,
  0xbb3bab1a, 0x6ca67c87, 0x5cc52c22, 0x3c030c60, 0x1c41edae, 0xfd8fcdec,
  0xad2abd0b, 0x8d689d49, 0x7e976eb6, 0x5ed54ef4, 0x2e321e51, 0x0e70ff9f,
  0xefbedfdd, 0xcffcbf1b, 0x9f598f78, 0x918881a9, 0xb1caa1eb, 0xd10cc12d,
  0xe16f1080, 0x00a130c2, 0x20e35004, 0x40257046, 0x83b99398, 0xa3fbb3da,
  0xc33dd31c, 0xe37ff35e, 0x129022f3, 0x32d24235, 0x52146277, 0x7256b5ea,
  0x95a88589, 0xf56ee54f, 0xd52cc50d, 0x34e224c3, 0x04817466, 0x64475424,
  0x4405a7db, 0xb7fa8799, 0xe75ff77e, 0xc71dd73c, 0x26d336f2, 0x069116b0,
  0x76764615, 0x5634d94c, 0xc96df90e, 0xe92f99c8, 0xb98aa9ab, 0x58444865,
  0x78066827, 0x18c008e1, 0x28a3cb7d, 0xdb5ceb3f, 0xfb1e8bf9, 0x9bd8abbb,
  0x4a755a54, 0x6a377a16, 0x0af11ad0, 0x2ab33a92, 0xed0fdd6c, 0xcd4dbdaa,
  0xad8b9de8, 0x8dc97c26, 0x5c644c45, 0x3ca22c83, 0x1ce00cc1, 0xef1fff3e,
  0xdf7caf9b, 0xbfba8fd9, 0x9ff86e17, 0x7e364e55, 0x2e933eb2, 0x0ed11ef0
+  }
};
```

```diff
int main(void)
{
  ...

  /* USER CODE BEGIN 2 */
  /* Configure LED4 and LED5 */
  BSP_LED_Init(LED4);
  BSP_LED_Init(LED5);

  /* Compute the CRC of "aDataBuffer" */
-  uwCRCValue = HAL_CRC_Calculate(&hcrc, (uint32_t *)aDataBuffer, BUFFER_SIZE);
+  uwCRCValue = HAL_CRC_Calculate(&hcrc, (uint32_t *)aDataBuffer.u32DataBuffer, BUFFER_SIZE);
```

From your side, you could initialise `u08DataBuffer` of the union, instead of `u32DataBuffer`, since you are using 8-bit arrays.

Doing this, you get no more warning. I hope this helps. Please note I did not push the test further to runtime.

With regards,